ROVI-org/auto-soh #2: Explore whether ProgPy is a good base

WardLT, 2024-06-13:
ProgPy is a code developed by NASA for fitting and deploying prognostic models of many different kinds of dynamical systems.

They already have a well-organized class structure for models of systems and state estimation algorithms, with several implementations of algorithms we will use in AutoSOH. 

Challenges:
- [ ] There is no differentiation between state parameters which represent the dynamic state of the system, and those which could capture long-term degradation. I'm exploring a solution in https://github.com/ROVI-org/auto-soh/tree/progpy-explore
- [ ] Class implementations use some practices that make modular codes difficult (e.g., overriding Python's serialization logic, relying heavily on class attributes). Starting to untangle them in https://github.com/ROVI-org/progpy/tree/allow-dynamic-parameters
- [ ] Output function for prognosis models lacks the control signal as an input. Attempted a PR in nasa/progpy#142

WardLT, 2024-06-17:
Pull request #3 mimicks ProgPy, but with some changes to how the base classes are implemented to better match how we've been approaching SOH.

I'll hope to close the delta between their base classes and ours enough that I can drop ours, but am moving forward with #3 to better understand our requirements.

victorventuri, 2024-07-10:
I think we have determined this not to be the case, right? @WardLT Can we close this issue?

WardLT, 2024-07-10:
I'd still like to keep it open. I closed that issue because ProgPy suggested a nice workaround.

I'm still keeping a mind open that we may be able to merge with ProgPy or PyBOP in the future, even though I think we should continue here to get a better of idea of what we need.

closed 2024-10-25 by WardLT