ROVI-org/auto-soh #7: Decide where the definition of which parameters are "state of health" should be

WardLT, 2024-06-21:
Right now, we list which parameters are being treated as state of health as part of the "SystemState" class.

However, many of our strategies for state estimation may rely on updating different parameters during different phases. Thus, it feels more likely that the list of which parameters to be updating should lie with the class that is using that information -- `OnlineEstimator`. 

If we do that, then the covariance matrix between each parameter should be an attribute of the OnlineEstimator as well because the State will not have knowledge of which parameters require covariances.

victorventuri, 2024-06-21:
I think that makes sense. I think we'll just need to make sure to provide an (optional?) argument to the several `to_numpy()` methods to list which `Field`s should be used when building the `np.ndarray` (which is something you partially have on your current version of `SystemState`)

WardLT, 2024-06-21:
Such a "get me only these variables" function would make a lot of sense for the `JointState`

victorventuri, 2024-07-22:
Should we close this? I think we reached a good place in the definition of `HealthVariable`

WardLT, 2024-07-22:
Yea, let's close

closed 2024-07-22 by WardLT