ROVI-org/auto-soh #8: Dealing with flexibility

victorventuri, 2024-06-24:
We want to allow the user to specify what type of ECM they want, as well as whether to include a series capacitor, and how many RC elements to consider, etc.. We also want to allow the user to specify which of these components should be estimated, and which should be considered fixed.

So, assume a user wants 3 RC components, but only wants us to estimate the RC values for 2 of them, as they are very certain of the values of the third one. How would we go about implementing something like this with `pydantic`'s `BaseModel`? 

My original idea was to dynamically add more RC components as needed (see [this issue](https://github.com/pydantic/pydantic/issues/1937)), but I feel this makes it difficult to deal with cases where some would be updated and some wouldn't...

victorventuri, 2024-06-24:
Perhaps the solution would be to create an RC_Elements class (based on the `HealthParameter` class) which holds all the RC elements, and its `to_numpy` method only outputs the internal parameters of those which can vary with time.

WardLT, 2024-06-26:
Have you already considered listing the RC elements as a tuple of floats? The to numpy function for the current SystemState supports attributes that are tuples of arbitrary length, which we could mirror in your version

WardLT, 2024-06-26:
One thought: the `SystemState` class currently differentiates the parameters within a list by appending a number after the name (e.g., `name_0`). https://github.com/ROVI-org/auto-soh/blob/pydantic-progpy/asoh/models/base.py#L75

Do you think we could have users provide either the name of the attribute to adjust all values (e.g., `'name'`) or a list of specific members within the list to get only get a subset (e.g., `'name_0'` as a string, or `('name', [0])` as a tuple)?

The list of which attributes to adjust would be part of the Estimator class in this picture, so that the same Parameter can be used with multiple filters each acting on different parameters.

PS. I deleted a comment I made earlier about the `tuple[float, ...]` because I misread your question. Sorry, I shouldn't go on GitHub _before_ coffee.

WardLT, 2024-07-05:
These idea have been incorporated into #11

closed 2024-07-05 by WardLT