ROVI-org/auto-soh #19: Decide a better name for the Python module

WardLT, 2024-07-05:
The name "auto-soh" didn't have much thought go into it and I at least want to the concept of "Advanced State of Health" (ASOH) in to emphasize that we're defining the health of a storage system many, nuanced parameters. 

A few ideas:

1. Just go to "Auto-ASOH", then name the modules "aasoh." However, I'm not sure how to pronounce that.
2. Something more metaphorical like "health bar," which isn't yet taken on PyPI (somehow!?)
3. Something esoteric like "Lachesis" or "Atropos," one of the [three Fates](https://en.wikipedia.org/wiki/Moirai). The library kind of deals with looking at the life history of a battery then judging its worth.

victorventuri, 2024-07-08:
One thing to keep in mind: will we integrate this with the forecasting models here, as part of the same package, or separately? To me, it makes a bit more sense to integrate them separately, but, if we want to do it all as part of the same package, that might impact our choice of name.

WardLT, 2024-07-08:
I'm not sure, but I do like the idea of integrating in the prognostics. That's making me feel Lachesis (or Decima, her Roman mirror) as an option. (I like metaphorical names for codes)

victorventuri, 2024-07-08:
Crazy idea as I am reading through the wikipedia link you shared: keep them separate, call the diagnostics one Clotho, the prognostics one, Lachesis, and the place where we integrate them all, Moerae. Need to find something to be Atropos...

victorventuri, 2024-07-23:
Or, maybe, the toolkit is Clotho, the diagnostics is Lachesis, and the prognosis is Atropos. Together, they form Moerae

victorventuri, 2024-07-23:
Sounds good! Message me when you merge these changes so I can pull them.

PR #37 (2024-07-23): Rename project to Moirae
Fixes #19
commit 194b2038 (2024-07-26): Rename project to Moirae (#37)

* Rename project to moirae

After the Three Fates

Fixes #19

* Move where we measure coverage

* Flake8 the correct dir
closed 2024-07-23 by WardLT