ROVI-org/auto-soh #24: Allow for batches of HealthVariable or transient states

WardLT, 2024-07-11:
It would be more convenient to treat an entire batch of HealthVariable or transient states as a single object, and this could open the door to performance optimizations (e.g., vector functions for updating the state).

A few ideas:
1. Allow updatable parameters in HealthVariable to be stored as 1 or 2d numpy arrays instead of floats or 1 d vectors. The additional dimension, if first, can be automatically treated as a batch dimension by numpy. 
2. Create a HealthVariableBatch class which maintains the values of each parameter as an internal list, then utility operations to get or all values of a certain parameter

victorventuri, 2024-07-23:
Just adding to the conversation here as I continue thinking about this issue, including things we've already discussed and problems we foresee: 
- if we want the `GeneralContainer` class to be able to store batches as well, it should still work seamless for cases where the batch cardinality is 1, and floats and 1D `numpy` arrays are passed
    - if we go this route, we will need to change a lot on how the `from_numpy()` method works, as well as other private functionality
- the `CellModel` still needs to work if we provide a single transient vector, single A-SOH, and single input/control object, as well as if a `JointStates` object is given. 
    - due to the fact that not all health variables are updatable, the `JointState` may not contain all the necessary info for the `CellModel`, so we also have to figure out a way to accommodate for that (like passing a `JointState` and an `AdvancedStateofHealth` together, and the model can fill in the `JointState` blanks with the ASOH object)
- if we try to also make the `HealthVariable` batchable, we may run into problems when dealing with the interpolation-style variables, seeing as the the interpolation would have to be re-done for each row in the 2D `base_values` field

victorventuri, 2024-07-23:
More thoughts about using the current classes (`GeneralContainer` and `HealthVariable`) as the containers of both single-instances of quantities, as well as batches of instances:
- part of the difficulty with the current classes is that, even for things that are like a "single instance" of a `GeneralContainer`  or `HealthVariable`, there are fields that are naturally a 1D numpy array. So, when we are batching, how will the objects and models know that an array is a batch of single-valued objects, or an actually single-instance iterable of something (without looking at the other fields)? For example, consider `i_rc` in `ECMTransientVector`, which is currently a `np.ndarray` containing as many values as there are RC components in the ECM. This also causes problems when updating from a numpy array, as it is critical to know the indices we should look at the new array, as well as the get the correct dimensionality. One work around I see for this is creating a new field, say `batch_size`, that specifies how many batches we are storing. I don't quite like that, as now the user also has to specify another thing, instead of just being able to provide the arrays with values. We could try to make it an `abstractproperty` instead (maybe?), and make sure all children have it? 
- for the `SOCInterpolatedHealth`, things can also become annoying, as, if the `base_values` are a 1D array, it generally assumes they have to be interpolated... but, what if that 1D array is simply a collection of single-values that need no interpolation? If it is a 2D array, then we need to provide a bunch of interpolating functions, one for each set of `base_values` (row) of the 2D array.
- I don't think all this necessarily  implies we cannot use the current classes as the holders of both batches and single-instances, but I do think it might require significant changes, including more field and model validators (which, hopefully, are not the cause for the inefficiencies in the joint estimation)

WardLT, 2024-07-23:
What if we accept scalars or 1-d arrays in the `__init__` function but then convert them to 2D such that any codes which uses the `GeneralContainer` or `HealthVariable` can safely assume they are always dealing with batched data?

victorventuri, 2024-07-24:
I believe that might work. But the conversion either needs to be done with a `field_validator`, or we will need to add a bunch of private attributes... Another thing is: if the user passes a 1D array of shape `(X,)` to, for example, `base_values`, even if we want to convert it to a 2D array, we need to know if it should be a `(X,1)` or a `(1,X)`, meaning, the values passed are for different batches, or are they needed for internal parametrization?

victorventuri, 2024-07-24:
If possible, I would prefer to not have to use `field_validator`s or `model_validator`s, since there is still a chance they are causing part of the inefficiencies in the estimators

WardLT, 2024-07-24:
> Another thing is: if the user passes a 1D array of shape (X,) to, for example, base_values, even if we want to convert it to a 2D array, we need to know if it should be a (X,1) or a (1,X)

I'm not sure. Maybe we just require that users cannot pass 1D arrays in areas it may be ambiguous. 

> If possible, I would prefer to not have to use field_validators or model_validators, since there is still a chance they are causing part of the inefficiencies in the estimators

Good point. I'm OK with them if they only run on creating the variable classes and not during update, as I think we can avoid running them often in that way.

victorventuri, 2024-07-24:
> I'm not sure. Maybe we just require that users cannot pass 1D arrays in areas it may be ambiguous.

Idk, I would feel pretty annoyed if I were trying to create a single health variable with a few parameters to interpolate, and the object raises an error complaining it wants a 2D array where the first dimension is the batch one. Maybe the best solution is to just add a `batch_size` field, default it to 1, make sure it is never marked as updatable, and use it when updating (and to make sure everything follows the same batch size)

> Good point. I'm OK with them if they only run on instantiating a new object and not during update, as I think we can avoid creating new objects often.

I believe a `model_validator` runs every time we change something to the model, while a `field_validator` only runs if we set `validate_assignment` to `True`, but it's worth a check.

WardLT, 2024-07-30:
Fixed by #40

closed 2024-07-30 by WardLT