ROVI-org/auto-soh #30: Streamline interface between models and estimators

victorventuri, 2024-07-18:
Currently, for every new model we create, we also need to create a corresponding `ModelFilterInterface` type class. In the future, when we introduce new models, we should revisit this, and considering writing a universal `ModelFilterInterface` that takes the model as input, and does all the same basic functionality regardless of the model used, or of the online estimator used.

WardLT, 2024-07-22:
We can also reduce the need to create as many subclasses more generally. There are a fair number of subclasses which redefine the docstring but don't provide new funcitonality (e.g., the [KalmanHiddenState](https://github.com/ROVI-org/auto-soh/blob/3d5d1a954da69ec9866ef560271bfd00e0eb1cdd/asoh/estimators/online/general/kalman/__init__.py#L6)) that would be a good target for eliminating as we prune subclasses of ModelFilterInterface

victorventuri, 2024-07-22:
Just going to point it out here based on the Slack discussion that the KalmanHiddenState is a child of two independent classes, so likely needs to exist on its own. However, I understand and agree with that sentiment, there are definitely a fair number of classes that are sort of like flair.

WardLT, 2024-07-31:
The ModelFilterInterface was eliminated in #38

closed 2024-07-31 by WardLT