ROVI-org/auto-soh #59: Joint UKF notebook shows numerical issues with Thenevin example

victorventuri, 2024-08-14:
When running the second example in the joint UKF demo notebook, numerical issues show up: somewhere along the way, a covariance matrix became not positive semi-definite, likely as a result of the large difference in order of magnitude between values in transient vectors and A-SOH parameters. 

We should investigate if re-adding the A-SOH normalization could help with this. 

This is something that, while less relevant, could be useful for the dual estimation as well, so maybe there is a way in which we can implement only once, and both frameworks have access to it. Tagging @WardLT to discuss further.

WardLT, 2024-08-14:
Do you think we can make a "NormalizerFilter" wrapper over a BaseFilter? 

I'm picturing a metaclass of BaseFilter which a filter it wraps and a list of normalization factors, then handles adjusting the hidden states?

victorventuri, 2024-08-14:
This is a great idea! I had not imagined the normalization sitting on the `filters` side of it, but that seems like it would be an easier way of making sure both joint and dual frameworks interact with it in the same way :thinking:

victorventuri, 2024-08-19:
@WardLT this is also happening in the dual, which is unexpected... I will do some more checks to see if the problem is not elsewhere, but perhaps we should consider prioritizing the normalization... will keep you posted

WardLT, 2024-08-21:
I was thinking about the normalization, and have a proposal: a wrapper over a filter which transforms data being passed to the underlying filter.

 The problem I missed before was that the filter makes calls to the physics model, and that we must denormalize before the points before calling the physics model. So, there must also be something to handle normalization for the model. The normalizing filter wrapper will also have a to apply a such a wrapper over the model of the underlying filter.

PS. This needs a picture. I'll sketch one tomorrow. Just wanted to drop this note while it was on the brain.

victorventuri, 2024-08-21:
As I am giving this some more thought, i will jot down ideas here. At first, there are two questions to consider:

1. Who needs to know that something (the A-SOH) is normalized?
2. Where does the normalization implementation live? 

I feel the answer to the first question guides the answer to the second question. Based on that, I discuss three candidate classes/objects that may need knowledge of normalization:

- `HealthVariable`: while i am open to discussion if you find this is viable, I am *a priori* strongly opposed to this because
    - Pros:
        - little to nothing on the estimator side would need to be changed
    - Cons: 
        - the model deals with "raw" values, and normalization is meaningless there
        -  similarly, on the model side, nothing should understand how to adapt filter-related quantities (like a covariance matrix) based on the normalization
        - I am of the opinion that this particular class is already doing a lot, so we should avoid as much as possible adding more stuff to it, especially if it can be added elsewhere

- Relevant `BaseCellWrapper`: to me, this is a more viable option
    - Pros:
        - intuitive (to me at least), as the wrapper is the natural model $\iff$ filter bridge 
    - Cons
        - the wrappers do not know what is the kind of filter being used; the don't posses knowledge of which `MultivariateRandomDistribution` is used, or how to (de-)normalize them
        - it could cause the wrappers to be bloated and maybe unreadable

- Some form of `FilterWrapper`: this is an interesting idea that we should consider in more detail
    - Pros:
        - given the numerical issues appear at the filter-level, it makes sense to try to address them there
        - given the filters know what kind of `MultivariateRandomDistribution` they deal with, they posses the knowledge to adequately convert means, covariances, etc. 
    - Cons:
       - the filters should not be able to distinguish between transient, A-SOH, etc., so they cannot be fully responsible for figuring out what to normalize and what to leave as "raw"
       - some form of pre-processing would be needed, as the filters shouldn't know what normalization vector to use, even if they are told to normalize (for instance, what to do in cases where the hysteresis parameter `H0` has `base_values` of zero?)

I like the idea of maintaining the `BaseCellWrapper`-type classes filter-agnostic, but, similarly, I like the idea of the filters being physics-agnostic, without knowledge of transients, A-SOH, etc. To me, one idea that is coming to mind is the `MultivariateRandomDistribution`: it can know if it is supposed to be normalized (and how), and does not need knowledge of what its dimensions represent. We would, however, have to alter is, as we would need different methods to `get_mean()` and `get_raw_mean()`... I don't really know if that would work, so let's see what we come up when we discuss this synchronously

WardLT, 2024-08-21:
These are great thoughts. Would you like to try defining an example interface of what you'd image for BaseCellWrapper having normalization? Doesn't have to be Python, but some psuedocode will be good. I'll do something similar for FilterWrapper and we can contrast

victorventuri, 2024-08-21:
I can try... but i am becoming less and less convinced that the `BaseCellWrapper` is the best approach... for some reason, I am particularly attracted to implementing that in the `MultivariateRandomDistribution` 😝

victorventuri, 2024-08-23:
Ok, so normalization did not solve this problem. After more analysis, I believe it stems from the UKF sigma points sampling illegal regions of A-SOH space, such as negative resistance values in the RC element, which blow up the current through that, and mess up all the numerics. It seems like that is the problem that needs fixing.

WardLT, 2024-08-24:
Interesting. I've had two thoughts so far for potential routes:

- make it so that the cell model clips values to stay within allowed ranges before running the physics (e.g., treating all negative values as zero).
- introduce a nonlinear coordinate conversion. We could have users provide a function and it's inverse for changing the points, and the Jacobian for both the function and inverse then use error propagation to assess the change in covariance.

WardLT, 2024-08-26:
The math I'm referencing for the "nonlinear coordinate conversion" is this - https://en.wikipedia.org/wiki/Propagation_of_uncertainty#Non-linear_combinations

Specifically, we'll need to change the `transform_covariance` to include the mean associated with a covariance, evaluate the Jacobian at that point, then use the Jacobian matrix to compute the new variance following the equation in that article. It's very similar to what we're doing with the linear transform except accounting for the fact that the derivatives of a nonlinear function are not constant at all points.

victorventuri, 2024-08-29:
Closing this as all numerical issues with UKF should now have been solved by normalization (PR https://github.com/ROVI-org/auto-soh/pull/75) and by non-linear transforms, such as absolute value (PR https://github.com/ROVI-org/auto-soh/pull/78)

closed 2024-08-29 by victorventuri