ROVI-org/auto-soh #60: Lack of a degradation model

victorventuri, 2024-08-14:
Currently, the code base does not poses the ability of degrading the A-SOH: there is no model that captures that, and the `OnlineEstimator`s assume that the A-SOH values remain the same from one time step to the next. 

It would be nice if we had a `DegradationModel` that could work in tandem with the estimators: it is called at every time step, and outputs new A-SOH parameters. It could, for the most part, only output the same values as the previous time step, but only after some number of effective cycles, or after some specified time, degrade the A-SOH parameters (it would have to keep its own history for that). 

Tagging @WardLT for additional discussions on (a) if this makes sense and (b) how to implement it if we decided to go for it.

WardLT, 2024-09-09:
That does make sense. I also like the places where you've noted that the degradation model will be used, like the model wrappers.

Building on your comment about "models keeping their own history," do you think we could use the time provided as part of the Inputs?

victorventuri, 2024-09-09:
I believe so; it could receive the same inputs as the `CellModel`, from `InputQuantities` to `TransientVector`s (to even `HealthVariable`s!) and figure out what to do with them (from storing, to computing effective cycles, to degrading the provide and/or stored A-SOH)

WardLT, 2024-09-09:
Maybe we could start with the time and then work on to more complex measures of age? E.g., should effective cycles be something fixed as an input or treated as a state which must be learned by the filter?

victorventuri, 2024-09-09:
I envision that anything that is derived should be part of the `DegradationModel` class. So, for example, it should be able to calculate effective cycles based on the history it receives.

WardLT, 2024-09-09:
I'm hoping that we can keep the Degradation Model stateless, like the CellModel. Shall we sync on what that could look like after doing a "time only" version?

victorventuri, 2024-09-09:
I don't know if it will be very easy to keep it "stateless", as it will probably need to at least have knowledge of the initial A-SOH of the battery (or the A-SOH at some point in time...) 

But here are a (un-ordered) list of things regarding this class:
- I will implement the base class in `moirae.models.base` and probably create a simple dummy example somewhere
- I will make it so it operates very similarly to the `CellModel` in terms of updating the A-SOH (that is, degrading it, based on either one `InputQuantities` object or a set of them)
- it will be used by the `Simulator` class, as well as the relevant `ModelWrappers` (`DegradationModelWrapper` and `JointCellModelWrapper` come to mind in particular)

WardLT, 2024-09-10:
Sounds great. Let's see how far we can get keeping an either stateless class. Attributes are fine, so long as they are read only.

Want to target an example where we include aging that depends only on time and not any other factors as a first PR?

victorventuri, 2024-09-10:
Yes, I will do that. I just need to do a few more investigations on another branch and I will switch over to this

closed 2024-09-12 by victorventuri