ROVI-org/auto-soh #68: Test batching/broadcasting with temperature dependence

victorventuri, 2024-08-19:
Some of the `SOCInterpolatedHealth` classes also have a temperature-dependent component (OCV, R0, etc.) While the temperature dependence is explicit, it needs to be properly checked with batching to ensure the broadcasting operations work. For example, in the OCV case, made up of reference OCV (OCV_r) and entropic OCV (OCV_e), we would need to check (when two or more things are batch, the batch size has to be the same!):

- OCV_r unbatched, OCV_e unbatched, SOC unbatched
- OCV_r unbatched, OCV_e unbatched, SOC batched
- OCV_r unbatched, OCV_e batched, SOC unbatched
- OCV_r unbatched, OCV_e batched, SOC batched
- OCV_r batched, OCV_e unbatched, SOC unbatched
- OCV_r batched, OCV_e unbatched, SOC batched
- OCV_r batched, OCV_e batched, SOC unbatched
- OCV_r batched, OCV_e batched, SOC batched

WardLT, 2024-11-04:
@victorventuri is this now complete with #96

victorventuri, 2024-11-04:
I don't think so.

#96 covers some of these cases for the resistance `HealthVariable`. Namely, all of the cases of base values and SOC being batched or un-batched are checked, but no temperature dependence is tested. 

#99 covers a few more indirectly, as the energy calculation requires calling the OCV, and that is checked with temperature dependence. However, it does so without batching the SOC. 

So, these two PRs together cover most of these cases, but we are still lacking a full test suite of the batching of transient, A-SOH, and temperature dependence all together.