ROVI-org/auto-soh #97: [BUG] Reading online estimates to HDF5 in wrong order

victorventuri, 2024-10-29:
**Describe the bug**
When reading online estimates that have been saved to an HDF5 file with
```python
for time, state_dist, output_dist in read_state_estimates('states.hdf5', per_timestep=True):
```
as in the [documentation](https://rovi-org.github.io/auto-soh/toolkit.html), it appears the state and output estimate are returned in the wrong order.

**To Reproduce**
Hard to do it here, as I found this out while looking at raw Kokam data, but the tldr is
1. Run online estimation and save the HDF% with `writer=HDF5Writer(hdf5_output=savefile, per_timestep='full')`
2. Read from the same file with `read_state_estimates(savefile, per_timestep=True)`

**Expected behavior**
In the documentation, it appears the `read_state_estimates` function would return an iterable of tuples (`time, state_estimate, output_estimate)`, but it seems like the last two are in the reverse order.

WardLT, 2024-10-29:
How sure are you? The example notebooks and unit tests seem to show the correct behavior: the state dists have all three state variables (soc, hyst, r0) and the output dista have just one (V)

victorventuri, 2024-10-29:
Not very, my computer also crashed while I was reading in the data (had the full raw datasets also loaded up, so ran out of memory). It is likely I was doing something wrong, so we can close this issue for now.

closed 2024-10-29 by victorventuri

WardLT, 2024-10-29:
👍 I noticed the per cycle version doesn't appear to be as well tested. Could that be where you saw problems?