ROVI-org/auto-soh #157: [FEATURE] Extractors require BatteryDataset objects

victorventuri, 2025-05-01:
**Is your feature request related to a problem? Please describe.**
In order to use the extractors, the data needs to be passed as a `BatteryDataset` object. I vaguely recall having discussed this before, but I think this complicates things unnecessary, as most `BatteryDataset`s contain multiple cycles, but, in many cases, one wishes to only extract values for a single cycle (such as a low C-rate capacity check or an HPPC). Additionally, in some cases (the capacity one, for instance), assumes the `'cycle_stats'` have already been computed from the raw data, which is not always the case (for example, the synthetic data we recently generated). 

**Describe the solution you'd like**
Ideally, all off the extractors would not take in arguments that are not defined in `moirae` already. Perhaps a happy middle-ground would be they take in a `pandas.DataFrame` that has been formatted according to the `battdat` standard, but not an actual `BatteryDataset` object. 

**Describe alternatives you've considered**
Saving every cycle as a separate `BatteryDataset`, which is inefficient.

**Additional context**
I think that, in general, as much as we can, we should avoid requiring arguments and objects to be used in `moirae`'s main functionalities, sticking instead to things that are defined in the package (some are obviously unavoidable, like numpy arrays). We can create a separate part of the package dedicated to establishing the proper communication channels (like what we currently have with the interface to do online estimation directly from raw data). This should also help in cases where external packages are updated and/or modified outside our control (fewer places where things break)

WardLT, 2025-05-02:
I'll keep pondering on this, but am unpersuaded about changing the interface. Using batterydataset:

- permits us to pass data / metadata in a standard format (e.g., if an extractor could use voltage limits specified in metadata)
- Integrates with analysis algorithms we may use in some extractors (e.g., for capacity estimation).
- Minimizes the number of things we create that are moirae-specific, keeping the focus tight on parameter estimation. 

Can we address your concerns now by making an easier path to chunking a battery datasets into smaller groups (e.g., by cycle)?

A second route is to add an "extract_from_raw" function to extractors which takes a dataframe as inputs and anything extracted from metadata as a kwarg. iirc, most of our extractors already have such a method. How about that?

victorventuri, 2025-05-02:
As discussed earlier today, I'm on board!

PR #162 (2025-05-06): Add method to extract from raw data
Adds a method to the Extractors which requires a dataframe of current, voltage, time and cycle number rather than a full dataset (Closes #157 )

Also fixes the OCV extractor by having it compute the columns it requires (fixes #158 )
closed 2025-05-06 by WardLT